DraftShift/StealthChanger #1: Holes too loose

viesturz, 2024-01-06:
Hey, I'm printing with quite spot-on dimensions, external perimenters first and the holes come out very loose.
The pins literally fall out and the bushings hold with minimal force.
The magnet holes are also quite oversized.

Also if you can update the instructions if they should be glued in place (recommended glue) or just press fitted.

viesturz, 2024-01-09:
Nope, just straight send to print. 
I'm tuning my flow based on top layer, typically resulting in 0.94 for ASA. And printing with external perimeters first which gives pretty spot on to slightly undersized external dimensions.

My guess would be that external perimeters last should add about 0.1-0.2mm to each wall and would mostly resolve this. 

Some pics for reference. 

[image]
[image]

TheSin-, 2024-01-09:
for ref I always print perim first, not for or against but this might be way I could never get lit working for me properly.

TheSin-, 2024-02-08:
I've tuned my printer and I'm now seeing the same issues.

I'm working on figuring out the best approach since we can't really have users drilling/reaming parts cause there is no direct access.

TheSin-, 2024-03-05:
This is fixed just not released yet.

closed 2024-03-14 by BT123D