DraftShift/StealthChanger #25: Stealthburner Heat Sets

24Tigger24, 2024-03-03:
The Stealthburner's heat sets are located on a radius (see left side in the picture), which makes it difficult to get them into the correct position when inserting them.

My suggestion: a flat surface (see right side in the picture) and the heat sets sit under the surface or on the surface with longer screws used
[image]

TheSin-, 2024-03-03:
It has to be on that side due to the amount of pull force required by the CW2 assembly to get a good amount of stiffness.  We also wanted to use the same sized fasteners, which is why they are where they are,  might be able to make a silghtly larger opening to get to them.  But all our efforts are currently on v2 which does not have this issue as all anymore since the geometry on v2 has been completely redone.

TheSin-, 2024-03-05:
actually after re reading I like this, if we do an other v1 maintenance release we will add this.  @BT123D

closed 2024-03-14 by BT123D