DraftShift/StealthChanger #32: Split shuttle into two parts and glue them later

whyme12, 2024-03-11:
Hi,
When printing the shuttle I hade huge issues. Loosing the small details for the belt due to support and slicer doing weird stuff.

I ended up printing the part sliced in half and gluing it back together after printing. 
- Its printed in ABS anyways and can easily be glued. 
- Both sides have perfect details. 
- The 4 Holes for the carriage can be used for Alignement

Only two downsides on the current model.
- The two holes on the bottom needed to be drilled after gluing
- there is no good plain to cut the part in half. So minor bridging/support is needed in the current model. 

Not sure about the clearance, but if the IR sensor thingy and the plane behind the middle bushing could be moved 0.4mm lower then the belt feature, then there would be a perfect plane to split the part in half.

Here is an example of how I printed the part:
[image]

Martorias, 2024-03-11:
It shouldn't be a big problem to print with supports, mine came out great. Most likely you'll use the shuttle keeper to hold the belt anyways.
Either way, there's a v2 in the works that won't be needing supports.

whyme12, 2024-03-11:
don't know what went wrong, but had huge issue printing the part in the recommended way. Never hade such issues before. 
Splitting the part made it fool proof in the end.

TheSin-, 2024-03-11:
splitting like this will cause LOTS of issues in the bushings. I would avoid this at all costs.  Once you are in a heated chamber I assume you will start to see other failures.

I'm closing this as I don't want to confuse others.

Having a properly tuned printing setup with tuned supports is a must.  We have builtin supports as a todo, so that's coming.

closed 2024-03-11 by TheSin-