DraftShift/StealthChanger #33: (Dragonburner carrier) Fasteners for pins and discord

cekim-git, 2024-03-23:
Discord invite returns "invalid" for me.  I was hoping to discuss this mod.

Added M3 fasteners for the pins for rigidity/adjustment and ease of installation (no super-glue).  This began as a work-around to make straight pins function while I waited for long-lead-time rounded pins, but I've kept it with my rounded pin for all the above.

Mods:
1. Added through and recess holes for M3 fasteners (M3x8SHCS for the top pins, M3x4FHCS)
2. Removed filleting in the bottom of the pin-hole (likely not needed - part of my experimentation to get straight pins and SS bushings I had on hand to work that I abandoned when rounded pins and brass bushings arrived and work much better).

Fusion360 and .step file

[StealthChanger_Dragonburner_Fastener_cekim.zip](https://github.com/Stealthchanger/Toolchanger/files/14733580/StealthChanger_Dragonburner_Fastener_cekim.zip)

BT123D, 2024-03-23:
Try this link

https://discord.gg/A5AUYTKU

TheSin-, 2024-03-23:
Since this can't be used for all backplates we will not include this, but this is addressed in v2 when it's ready, sine there is no PR and it looks like you used older version I'm just going to close this to not confuse other users.

closed 2024-03-23 by TheSin-