DraftShift/StealthChanger #37: Build of material/BOM

julianvolden, 2024-03-26:
The BOM needs to be updated..
Does not tell you to order 3x12mm bhsc if planning to use the shuttlekeeper for the SB. 

should also be updated with what printed parts that you need for the BOM option chosen.

my hotend is now dismantled, and obviously i need the belthelper to make it easier to install,and that stl was hidden under "usermod" and not STL. 
so it was a 2step forward, 1step backward operstion for me 😅

TheSin-, 2024-03-26:
Sorry to here this, the belt helper is just a user tool to help it's not technically required, just like for stock Voron there are such tools as well but they aren't part of the BOM.

As for the bolts they are listed in the manual as it's an optional part.  We will see about adding them to the BOM

TheSin-, 2024-03-26:
Thank you I have updated the docs

closed 2024-03-26 by TheSin-