DraftShift/StealthChanger #39: Shuttle keeper is too loose

Ficik, 2024-03-30:
I needed to reprint the shuttle keeper and I noticed that I can't tention my belts, they're slipping thru the keeper.
I found out that commit `73256c1 6mm Shuttle Keeper fix` offset shuttle keeper from mgn rail by 1mm, I agree that original 0.2mm was bit too tight for gt2, but 1.2mm can fit almost whole belt including teeth (1.38mm). 

Since belt thickness without teeth is 0.63mm, spacing should be around that value (0.6mm).

Top view of the rail carrage with shuttle keeper. Blue is pre patch, black is post patch

[image]

TheSin-, 2024-03-30:
@BT123D

BT123D, 2024-03-30:
Can you show any evidence of this? I've just tested with mine and this is not the case. 

The belt grooves are 0.6mm away from the metal face of the carriage to allow the 0.63mm of a gt2 belt, and they're 1mm away from the wipers. This is to account to variances in rails and because the belt should not be pressing on the wiper as this will cause issues

Ficik, 2024-03-31:
First I, need to apologize, I didn't have right cad model of mgn12h, so the image above is bit missleading it's mgn12c, length of the carriage is wrong. Here's new drawing:

[image]

As I was looking for proper dimensional drawing of mgn rail, I noticed that height of the bump of metal block body (or more precisely the depth of the valley between them) is not specified, neither is the height of green wipers and vary quite significantly between manufacturers. This probably explains why my belts slip and yours doesn't.

Anyway If you mean by wipers the whole green part then the only part that should clamp the belt in place is this highlighted one:

[image]

Which I'll argue that is rather small and bit too far from closest tooth (the one inside the hole) and that you actually are clamping the belt against the green wipers, mine are just bit lower than yours. After all almost all the teeth are above that part.

[image]

Step file of keeper on carriage: https://drive.google.com/file/d/1qZrFI_c3-GE3vv6t1esehQFsj6f-Y4Fj/view?usp=sharing

Step file of tweaked keeper: https://drive.google.com/file/d/1lZN_4F2ZRK3eo2AGeK-WC1YBED-E2koW/view?usp=sharing

BT123D, 2024-03-31:
Thank you for including the fixed step file for this. Sorry the current part didn't work for you but I'm happy that you could mod the keeper to work. Tomorrow I'll update to repo to suit your fix

TheSin-, 2024-04-01:
I have confirmed this issue for 6mm belts.

closed 2024-05-25 by BT123D