DraftShift/StealthChanger #53: Optional spine screws

AstronIsTaken, 2024-06-25:
Out of curiosity (since I'm printing new v1.1.1(?) shuttle) I tried to add spine screws to v1.1 shuttle and, unfortunately, shuttle cracked in between layers.  2nd screw hole I predrilled with M3 reamer almost all the way and screw went in flawlessly.

It might be hard to achieve perfect tolerances for such a long horizontally printed screw holes, so it might be good idea to add a note to guide to use reamer  before adding spine screws or increase screw holes diameter a bit.

TheSin-, 2024-06-27:
this is a good point thought oddly the first time it's been brought up, I'll look into it.

AstronIsTaken, 2024-06-27:
It might mean feature is not used at all you can just retire it. Or my tolerances are so bad or I had block stuck inside hole, so shuttle cracked

TheSin-, 2024-06-27:
Oh I know it's used by many as we talk about it, that being said shuttles aren't printed as much as tool plates so once it's working....

kiwidave72, 2024-06-29:
ive had a few issues, than i reprinted and reamer'ed out some plastic with a reamer until i could get the bushings in without craking. think this is more of a case by case thing as my printer isnt the best turned one in the world.

john-f-chamberlain, 2024-06-29:
Did you use fully threaded screws or are they only partially threaded? When added the spine screws to mine I noticed the ABS discoloring due to stress caused by the partially threaded section being slightly wider than the threaded section. I would imagine under certain circumstances this could add enough stress to cause the part to split.

TheSin-, 2024-06-29:
full thread

AstronIsTaken, 2024-06-29:
In my case I had partially threaded

TheSin-, 2024-06-29:
maybe we will change the document to say fully threaded then.  The issue is that you want it to be tight the entire length, I could open it half way for a shoulder but then the bottom wouldn't benefit from the spine as much.  And the bottom needs it the most.

Altirix, 2024-07-04:
yeah i actually had this happen to me with v1 because i used partially threadded screws on my first go. had way more of them. rummaged thru my misc unsorted screws and found fully threaded which were fine.

TheSin-, 2024-07-04:
Added notes to use fully threaded https://github.com/DraftShift/StealthChanger/commit/f7f19c9c4d7fe11cdcacf678c8a3aee60b2f2538

closed 2024-07-04 by TheSin-