DraftShift/StealthChanger #55: Dragonburner_Spacer.step

leesafur, 2024-07-13:
Please add the Dragonburner Spacer step file to the cad folder. All spacers seem to be missing.

TheSin-, 2024-07-13:
It should be in the dragonburner backplate step already, I’ll check in the morning

TheSin-, 2024-07-13:
Confirmed the spacer is in the backpay file where it should be.

Closing ticket as it's invalid

closed 2024-07-13 by TheSin-

leesafur, 2024-07-13:
This is where i'm looking CAD/Backplates/DragonBurner.step and its not in there. However I did find it in one of the forks that hasint been updated.

TheSin-, 2024-07-13:
I just opened that file directly from the repo it and it’s right there as it opens

leesafur, 2024-07-13:
This one has the spacer https://github.com/DraftShift/StealthChanger/blob/main/Legacy/V1/CAD/Backplates/StealthChanger_Dragonburner.step
This one is the newer version where the tap standoff is longer but theres no spacer. Will the V1 spacer work, it seems to fit fine. https://github.com/DraftShift/StealthChanger/blob/main/CAD/Backplates/DragonBurner.step

TheSin-, 2024-07-13:
Yes spacer wasn’t changed but it’s in the v1.1 step file