DraftShift/StealthChanger #74: Trouble with QGL

sgermain06, 2024-09-22:
After printing everything I need for my Voron 2.4 350, following guides and all, ever since I installed the shuttle for stealthchanger and installed my dragonburner on it, I'm having some really weird issues with performing QGL. For some reason, the variations are all over the place. The toolhead seems to fit perfectly on the shuttle, it moves up and down freely, it doesn't feel loose or anything. Also, I have minimal variation when performing a PROBE_ACCURACY (usually under 0.003 variation) at all 4 corners. At first, I thought that maybe some of my Z steppers were out of order or something, but using the STEPPER_BUZZ command, I confirmed that they are all configured correctly.

Before I switched to this new setup, I had a single stealthburner with VoronTap and EBB2209 canbus board on Klipper v0.11 and had zero issues with QGL. I was also using all the same hardware, minus the Dragonburner and the EBB36.

Here's what I'm running:
Voron 2.4 350mm
Stealthchanger shuttle
Dragonburner toolhead (x2)
BTT Octopus Pro v1.1 (H723)
BTT EBB36 canbus breakout boards (x2)
BTT U2C

klipper - v0.12.0-302-g87ac693
klipper-toolchanger - v0.0.81-ge53f4233-inferred
moonraker - v0.9.0-g71f9e677
Raspberry Pi 4+ 8gb with Raspbian bullseye, fully updated

If anyone has any suggestion on how I could troubleshoot this, I'm all ears!

TheSin-, 2024-09-22:
98% of the time this is bad umbilicals or your mags are proud and you aren’t fully seated.  You should not see and pin or bushing at all when fully seated and you umbilicals should not push/pull on the tool at all you need a nice arc

if you are unsure please post pics but one of those will most likely fix everything

sgermain06, 2024-09-23:
I figured it out, well @techn0phreak figured it out lol! I had a Z-axis pulley that the set screw was loose (not sure how it happened, but it happened). I tightened everything back up, added some blue loctite and that thing is super solid now, QGL is going through without a problem.

Thanks for your suggestions!

closed 2024-09-23 by sgermain06