DraftShift/StealthChanger #103: Step and STL for Hall_Effect_Shuttlekeeper are not in sync

KlausHans, 2025-02-26:
I did a little bit of remixing when i realized that the step files and the stls are different for the Hall_Effect_Shuttlekeeper:

[image]

That begs the question, are the stls or the step outdated?
Thanks!

PS: I don't know if other models are affected by the same issue.

TheSin-, 2025-02-26:
use web version

closed 2025-02-26 by TheSin-

KlausHans, 2025-02-26:
What do you mean? Where do i find a link? Am i blind?

TheSin-, 2025-02-26:
use the STL files they are newest, it's LOTS of work to extract all the different steps, so for small changes we don't always do it.  STLs are ALWAYS the safest bet.  Next time there is a larger change we will update the STEPs