DraftShift/StealthChanger #120: Document mentions unavailable assets.

aeiounix, 2025-05-03:
https://github.com/DraftShift/StealthChanger/tree/bef00555b1e31180e6e3ea8c880e65f3fc1085d8/UserMods/traxman25#dragon-burner-ebb36-backing-plate

In the above link @traxman25 mentions DragonBurner_EBB36_OrbiterV2_mounts.STEP and a .3mf file which are not present in the repository.

TheSin-, 2025-05-03:
That’s a usermod, please contact the user on discord to report it

closed 2025-05-03 by TheSin-

traxman25, 2025-05-05:
> https://github.com/DraftShift/StealthChanger/tree/bef00555b1e31180e6e3ea8c880e65f3fc1085d8/UserMods/traxman25#dragon-burner-ebb36-backing-plate
>
> In the above link [...]

The mount is in the 3mf located here: https://github.com/DraftShift/StealthChanger/blob/bef00555b1e31180e6e3ea8c880e65f3fc1085d8/UserMods/traxman25/Dragonburner_EBB36_SC_Mount/V1.0/DragonBurner_EBB36_OrbiterV2_mounts%20V1.0.3mf

aeiounix, 2025-05-05:
Yeeaaah. I found it. Silly really. Thank you.